lexus2k/lcdgfx #78: New Hardware documentation

buelowp, 2022-01-07:
I would like to contribute a port to the Particle build system. Is there documentation somewhere, or can we work some documentation out, for how to add a new folder in lcd_hal for particle? I expect it wouldn't be too difficult.

I've forked the repo for the time being so I can hack away at it when I have a bit of time. If I can get it quickly enough, we can talk about the additional documentation which I would be happy to help with. It would be your choice to include said support in the original repo I suppose.

My goal is to support the SH1107 Featherwing from Adafruit for use with the Particle Argon for a project I'm getting started.

Thank you.

lexus2k, 2022-01-09:
Hi,

I'm not familiar to the Particle build system?
Are you talking about new hardware platform support, or about just integrating with new build system?
As far as I understand, Argon has nRF52840 onboard as main SoC, and from the documentation (https://docs.particle.io/reference/device-os/firmware) I see that it has Arduino-like API.
So, could you please describe in details, what exactly is required to be changed in the library to make it compatible with the Argon Particle build system?
If you have ready-to use patch, can you publish it to shed the light on the required changes?

Thank you.

buelowp, 2022-01-09:
I'm just getting started, and don't have a functional patch just yet. I also do not have a lot of details on what needs to be changed, but I'm working on it. No luck just yet. 

In fact, it is both, a new HW platform (Particle) and they have a custom FreeRTOS based build system that presents as Arduino compatible. So, I need to get both going at the same time, though for the time being, my goal is to only to support the FeatherWing OLED solution on the Argon. There isn't a good reason why once in place, this lib couldn't be used with the Photon or Boron as well and the variety of OLED devices the library supports. The challenge I face is that the SH1107 is not well supported in this space, and no current library (lcdgfx, u8g2, Adafruit_GFX) that does support the SH1107 builds for Particle devices. The following commits to my repo are my first pass at what I think I needed to get started on the integration side. I'll post the most recent build issues below. I plan to get back to this tomorrow or Tuesday as time permits and see what progress I can make.

https://github.com/buelowp/lcdgfx/commit/f35b7cc6368132221c1d4137cac9fc9771726476
https://github.com/buelowp/lcdgfx/commit/59a889d69fdee0339f789cb9f92ebacfd036d261
https://github.com/buelowp/lcdgfx/commit/6b9da6c1b2e874ae85d3a2af14ce3730bef827bc

I wasn't able to do much this weekend, so the current error is probably stupid, but I haven't had time to look into it. If you can tell me if I'm on the right track at least, that might help. I will keep hacking away at it, and see what I can do.

Yes, Particle is Arduino compatible and the Argon is an nRF chip. Most Arduino libraries tend to work, but he more complicated ones generally need some help. I've done a few ports before, and there are others who do the same (https://github.com/rickkas7). I was a bit surprised the Adafruit library isn't supported yet, but I think I like your system better, so I'm going to try to get this working first. 

Generically, Wire exists, and my idea was just to copy the Arduino hal in lcd_hal to a new particle folder, as it *should* work as is, minus some include files. I've been bumping around with that for a bit yesterday, and got as far as below. Whatever I did isn't quite right though, as it's mixing C and C++ somehow. I'll need to sort that out.

You can find the specific device I'm trying to use here
https://www.adafruit.com/product/4650
And the datasheet is just the SH1107 datasheet, but linked here just in case
https://cdn-learn.adafruit.com/assets/assets/000/094/580/original/SH1107_datasheet.pdf?1599165309

Thanks for any help if you can give it.

Current status
Latest build error
```
mkdir -p ../build/target/user/platform-12-m/mylcdgfx/lcdgfx/src/v2/
/home/pete/.particle/toolchains/gcc-arm/10.2.1/bin/arm-none-eabi-gcc -DSTM32_DEVICE -DnRF52840 -DNRF52840_XXAA -DPLATFORM_THREADING=1 -DPLATFORM_ID=12 -DPLATFORM_NAME=argon -DPLATFORM_GEN=3 -DUSBD_VID_SPARK=0x2B04 -DUSBD_PID_DFU=0xD00C -DUSBD_PID_CDC=0xC00C -DSPARK_PLATFORM -g3 -gdwarf-2 -Os -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfloat-abi=hard -mfpu=fpv4-sp-d16 --specs=nano.specs -DINCLUDE_PLATFORM=1 -DSOFTDEVICE_PRESENT=1 -DS140 -DPRODUCT_ID=12 -DPRODUCT_FIRMWARE_VERSION=65535 -D_WIZCHIP_=W5500 -DUSE_STDPERIPH_DRIVER -DDFU_BUILD_ENABLE -DLFS_CONFIG=lfs_config.h -DSYSTEM_VERSION_STRING=3.1.0 -DRELEASE_BUILD -I./inc -I../wiring/inc -I../system/inc -I../services/inc -I../third_party/nanopb/nanopb -I../communication/inc -I../hal/inc -I../hal/shared -I../hal/src/argon -I../hal/network/lwip/esp32 -I../hal/network/ncp/wifi -I../hal/src/nRF52840 -I../hal/src/armv7 -I../hal/src/nRF52840 -I../hal/src/nRF52840/lwip -I../hal/src/nRF52840/freertos -I../hal/src/nRF52840/mbedtls -I../hal/src/nRF52840/littlefs -I../hal -I../hal/src/nRF52840/posix -I../hal/network -I../hal/network/api -I../hal/network/lwip -I../hal/network/lwip/posix -I../hal/network/lwip/wiznet -I../hal/network/ncp -I../hal/network/ncp/at_parser -I../hal/network/util -I../third_party/lwip/lwip/src/include -I../third_party/freertos/freertos/FreeRTOS/Source/include -I../third_party/nrf5_sdk/nrf5_sdk/external/freertos/portable/GCC/nrf52 -I../third_party/nrf5_sdk/nrf5_sdk/external/freertos/portable/CMSIS/nrf52 -I../third_party/wiznet_driver/wiznet_driver/Ethernet -I../gsm0710muxer/gsm0710muxer/include -I../platform/shared/inc -I../third_party/nrf5_sdk -I../third_party/nrf5_sdk/nrf5_sdk/components/toolchain/cmsis/include -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/drivers -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/drivers/include -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/hal -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/mdk -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/hal -I../third_party/nrf5_sdk/nrf5_sdk/modules/nrfx/drivers/src/prs -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/util -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140/headers -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/s140/headers/nrf52 -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/common -I../third_party/nrf5_sdk/nrf5_sdk/components/softdevice/mbr/nrf52840/headers -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/fstorage -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/experimental_section_vars -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/atomic -I../third_party/nrf5_sdk/nrf5_sdk/components/libraries/crc32 -I../third_party/nrf5_sdk/nrf5_sdk/integration/nrfx/legacy  [... cut ...]
                 from /home/pete/Projects/mylcdgfx/lib/lcdgfx/src/v2/utf8.c:29:
./inc/application.h:33:10: fatal error: chrono: No such file or directory
   33 | #include <chrono>
      |          ^~~~~~~~
compilation terminated.
```
Current source
```
#include "lcdgfx.h"
DisplaySH1107_128x64_I2C display(-1);

// setup() runs once, when the device is first turned on.
void setup() 
{
    display.begin();
    display.setFixedFont(ssd1306xled_font6x8);

    display.fill( 0x00 );
}

// loop() runs over and over again, as quickly as it can execute.
void loop() 
{
}
```

lexus2k, 2022-01-10:
Since the authors of the Particle system claim, that they support Arduino libraries, the lcdgfx doesn't need heavy changes. However, it is still required from the Particle system to support `*.inl` files, when uploading library.
For some reason, `whitelist` in library properties doesn't work as expected (per the documentation).
If you have the solution for the `whitelist` field, let me know

buelowp, 2022-01-10:
Where you able to get it to build? I can ask on the community about whitelisting, but I'd be curious to know how you got it to build and what I did wrong?

lexus2k, 2022-01-10:
I didn't build it because `particle library upload` doesn't upload *.inl files, needed for the compilation. And adding `whitelist=*.inl` to the library.properties doesn't solve the problem.

buelowp, 2022-01-10:
You can build against a library without uploading. The challenge with uploading libraries to particle is that it consumes the library namespace, and it's hard to clean that up. Once it's ready to publish, it will consume that namespace forever if anyone uses it as a dependency. That's fine for public libs, you don't want two with the same name. Private libs don't consume public namespace, but it's still a bit of a pain if you need to change something dramatic in your local library namespace. 

A better option if you have time, would be to use the vscode project creation to create a test project.

Then just add your library to the project/lib directory (symlink works great in Linux/Mac). Then update the project.properties to add
```
dependency.lcdgfx=1.1.1
```
A new project setup this way will allow to build your setup with a local library. No upload or library changes needed to make sure it works first.

The first big issue you will face is PROGMEM isn't defined for src/canvas/fonts/fonts.h which is easy to fix
```
#ifndef PROGMEM
#define PROGMEM
#endif
```
although I suspect there is a more elegant fix that I haven't gotten to yet. Then, there is no avr/interrupt.h available, so it fails in io.h then when you simply #include <Arduino.h> at the top of your .ino file. I'm still researching that one.

I'd be happy to get you the parts to test this if you are willing to add support. I'm going to keep working on it for now too, but will get sidetracked this week as I actually will have to go into the office some and as a result, will have less time for coding at home due to commute and kids.

What was the error you got uploading? I haven't tried just yet as I'm not ready to consume that library name. I will ask on the community board about .inl files and uploads when I can.

lexus2k, 2022-01-10:
> A new project setup this way will allow to build your setup with a local library. No upload or library changes needed to make sure it works first.

The problem here is that `particle compile argon` doesn't work locally. Even if the library is placed in the project subdir `lib`, particle cli uploads the local sources to the cloud, and initiates compilation on the cloud, not on the local PC.

> to use the vscode project creation to create a test project.

To make the build system to compile the project locally it is required to recreate all build commands calls and all required defines manually. I don't think that this is the right way.

If the `whitelist` worked, then the lcdgfx library could be integrated to particle in half an hour.

buelowp, 2022-01-10:
Let me work on whitelisting then. I'll let you know.

buelowp, 2022-01-13:
Ok, the .inl problem is related to some code directly in the particle binary. Nothing to do with the library.properties. I guess for now, it's just what it is. I'll keep trying to do it locally and see where I get I suppose.

buelowp, 2022-01-13:
OK, after much haggling with this, the display is still rotated and offset. I don't seem to be able to make this display start in the right place. I'm OK if you close this. I did fork and my fork has the changes to compile and run, even if it doesn't look very good. Maybe someone else can figure out why it's offset so weird.

closed 2026-04-03 by lexus2k