lexus2k/lcdgfx #79: ESP8266 and ESP32 generic module connections

star297, 2022-01-16:
I would love to use this but does anyone know how to connect it up?

Platform: ESP32 and ESP8266 generic modules.
Arduino v1.8.13
Display: SSD1331 oled display(96x64) SPI mode.

This makes sense to me and works with my library on Mbed, I set my device up using this:
```ssd1331 oled(D8, D7, D10, D11, NC, D13); // cs, res, dc, SDA(mosi), (miso nc), sck  (STM Arduino pin numbers)```

But I don't understand this:

```DisplaySSD1331_96x64x16_SPI display(4,{-1, -1, 5, 0,-1,-1});```
```// Use this line for ESP8266 Arduino style rst=4, CS=-1, DC=5```
```// And ESP8266 RTOS IDF. GPIO4 is D2, GPIO5 is D1 on NodeMCU boards```

``` ESP8266: connect LCD to D1(D/C), D2(CS), RX(RES), D7(DIN), D5(CLK)```

Do I specify the pin numbers here, if so which pins are for what function?
````(4,{-1, -1, 5, 0,-1,-1}````
I assume the -1 means NC (no connection) and the rest are the data connections?
What does the '4' mean?
ESP uses GPIO pin numbers, which pin numbers do I use and can I specify alternative pins?

Many thanks

lexus2k, 2022-01-16:
According to the [documentation](https://codedocs.xyz/lexus2k/lcdgfx) DisplaySSD1331_96x64x16_SPI constructor has 2 parameters:
<html><body>

rstPin | pin controlling LCD reset (-1 if not used)
-- | --
config | platform spi configuration. Please refer to SPlatformSpiConfig.

</body>
</html>

If you refer to SPlatformSpiConfig description: https://codedocs.xyz/lexus2k/lcdgfx/struct_s_platform_spi_config.html, you will find information on each structure field. 
Since default busId is VSPI_HOST, then you just need to use SPI pins related to VSPI_HOST according to the platform you have.
You may specify exact pin numbers, you want to use, just check agains your dev.module pinout.

lexus2k, 2022-01-20:
Hi,

please check the latest master branch. It contains some changes for SPI displays.

lexus2k, 2022-02-14:
Feel free to reopen

closed 2022-02-14 by lexus2k

star297, 2022-02-15:
Thank you but I don't think you understand what I'm asking.

```ESP uses GPIO pin numbers, which pin numbers do I use and can I specify alternative pins?```

The ESP32 has flexible GPIO pin defines, so I can use any pin combination for 'most' functions.
I want to set the GPIO pins of my Generic ESP32 module to the connect to the display.

```ESP8266: connect LCD to D1(D/C), D2(CS), RX(RES), D7(DIN), D5(CLK)```
how does that corelate to:
```DisplaySSD1331_96x64x8_SPI display(4,{-1, -1, 5, 0,-1,-1});```

Guessing at what you are indicating above,
```What does the '4' mean?```
Is it the RST pin?

Without trawling through loads of documentation is it possible for you to give me a simple example like this Adafruit example?

```
// LCD, any pins can be used
#define SHARP_SCK  14
#define SHARP_MOSI 26
#define SHARP_SS   12
// Set the size of the display here, e.g. 400x240
Adafruit_SharpMem display(SHARP_SCK, SHARP_MOSI, SHARP_SS, 400, 240);
```

Or do I override the default pins and also set SPI frequency in:
```display.begin();```

lexus2k, 2022-02-17:
> ESP uses GPIO pin numbers, which pin numbers do I use and can I specify alternative pins?

It depends on the hardware you use. So, yes, you can specify alternative pins.

> ESP8266: connect LCD to D1(D/C), D2(CS), RX(RES), D7(DIN), D5(CLK)
> how does that corelate to:
> DisplaySSD1331_96x64x8_SPI display(4,{-1, -1, 5, 0,-1,-1});

From my post above: please refer to the documentation. it describes the fields of SPlatformSpiConfig structure.
According to the [documentation](https://codedocs.xyz/lexus2k/lcdgfx) DisplaySSD1331_96x64x16_SPI constructor has 2 parameters:
<html><body>

rstPin | pin controlling LCD reset (-1 if not used)
-- | --
config | platform spi configuration. Please refer to SPlatformSpiConfig.

</body>
</html>

If you refer to SPlatformSpiConfig description: https://codedocs.xyz/lexus2k/lcdgfx/struct_s_platform_spi_config.html, you will find information on each structure field. 

> Without trawling through loads of documentation is it possible for you to give me a simple example like this Adafruit example?

I pointed the exact link to the information you need above. And here is below part of ssd1331_demo code:

```.cpp
// The parameters are  RST pin, BUS number, CS pin, DC pin, FREQ (0 means default), CLK pin, MOSI pin
DisplaySSD1331_96x64x8_SPI display(3,{-1, 4, 5, 0,-1,-1}); // Use this line for Atmega328p
//DisplaySSD1331_96x64x8_SPI display(4,{-1, -1, 5, 0,-1,-1});  // Use this line for ESP8266 Arduino style rst=4, CS=-1, DC=5
                                                             // And ESP8266 RTOS IDF. GPIO4 is D2, GPIO5 is D1 on NodeMCU boards
```

star297, 2022-02-17:
Found you had a ESP32 set up here:
https://github.com/lexus2k/lcdgfx/blob/master/examples/demos/ssd1351_demo/ssd1351_demo.ino

Used this in the ssd1331 8 and 16bit demo and is working.
```
//  ESP32: connect LCD  gpio23(DIN/SDA), gpio18(CLK)                                                            
DisplaySSD1331_96x64x16_SPI display(22,{-1, 5, 21, 0,-1,-1}); // Use this line for ESP32 (VSPI)  (gpio22=RST, gpio5=CE for VSPI, gpio21=D/C
```

> {-1, 5, 21, 0,-1,-1}

I assume this translates to:

SPI channel, (-1) = default SPI0 (VSPI)
CE or CS = 5
DC = 21
SPI frequency (0) = (platform or display maximum default) 
SPI CLK  (-1) = platform default (18)
SPI MOSI (-1) = platform default(23)

Many thanks