lexus2k/lcdgfx #81: 'SPI_DMA_DISABLED' was not declared

KaiJiao, 2022-02-11:
**Describe the bug**
C:\Users\jiao\Documents\Arduino\libraries\lcdgfx\src\lcd_hal\esp\esp32_spi.cpp: In member function 'void EspSpi::begin()':
C:\Users\jiao\Documents\Arduino\libraries\lcdgfx\src\lcd_hal\esp\esp32_spi.cpp:90:66: error: 'SPI_DMA_DISABLED' was not declared in this scope:
spi_bus_initialize(m_busId ? VSPI_HOST : HSPI_HOST, &buscfg, SPI_DMA_DISABLED); // 0 -no dma

When I verify an empty code which just include the lcdgfx library in Arduino IDE, it shows this error. 

Do you know how can I modify it? Thank you very much!

Kai

[image]

lexus2k, 2022-02-11:
Hi, just replace SPI_DMA_DISABLED with 0.

This is inconsistency in different SDK versions.

lexus2k, 2022-02-13:
Le me know if it works for you.

Also it would be nice to get know the version of the tools and the board you're using.

KaiJiao, 2022-02-14:
Sorry for my late reply, yes that's work for me! Thank you very much!

lexus2k, 2022-02-14:
To prevent such issue in the future, I have updated the code to use 0 on the latest master branch.

closed 2022-02-14 by lexus2k

CNSNGopenSourceProjects, 2022-02-15:
It should be nice if to publish this library fix for Arduino IDE.
The bug fix work for me.
Thank you.

lexus2k, 2022-02-15:
Done. v1.1.4 will be available soon from Arduino IDE.

commit b964bdf3 (2022-02-14): Compilation fix #81