lexus2k/lcdgfx #83: ST7735 Incorrect color order BGR/RGB

thumptech, 2022-03-24:
**Describe the bug**
The colors are incorrect when using the DisplayST7735_128x128x16_SPI, probably all ST7735 drivers

**To Reproduce**
[I just used the driver on this module](http://www.lcdwiki.com/1.44inch_Arduino_SPI_Module_ST7735S_SKU:MAR1441)

**Expected behavior**
The correct colors should be displayed

**Please complete the following information:**
 - library version 1.14
 - LCD display type ST7735_128x128
 - Platform ESP-IDF ESP32

**Additional context**
Changing line 174 of lcd_st7735.inl to:
`0x36, 0x01, 0b00001000, // MADCTL`
sets the correct bit order and fixes the problem for me by telling the controller to use RGB and not BGR

If this is specific to the board I have it would be nice to have a function in the driver to set the pixel format.

lexus2k, 2022-03-26:
Please, find the fix in the latest commit 2de202c5123b8d18dee90674e21ffccb3a5552e7.
Let me know the test results

```.cpp
    display.getInterface().setRgbMode(1); // or 0
```

thumptech, 2022-03-30:
Apologies, i have been away from work due to illness and haven't been back
to test this. Thankyou for your prompt attention to this matter.

thumptech, 2022-03-31:
I've tested the setRgbMode() function and it is working as expected. Thankyou.

closed 2022-03-31 by thumptech

lexus2k, 2022-03-31:
You're welcome. Once enough new features to publish are ready, the new version of the library will be released for the Arduino Library Manager.

thumptech, 2022-03-31:
Thankyou. I just checkout the repo into my ESP-IDF project so it works fine
right now. There are other problems with this display, but ill come back to
that and create a new issue once ive investigated it.