lexus2k/lcdgfx #86: ST7789 240x240 display

Adrianotiger, 2022-06-10:
Hi,
I was trying to add the 240x240 display to the list, but I can't see how I should do it.

If I just add the template an offset to the image is set.

Probably I need to edit some data here:
```

static const PROGMEM uint8_t s_ST7789_lcd135x240x16_initData[] = {
#ifdef SDL_EMULATION
    SDL_LCD_ST7789, 0x00, 0b00010000, 0x00,
#endif
    0x01, CMD_DELAY, 150,                     // SWRESET sw reset. not needed, we do hardware reset
    0x11, CMD_DELAY, 20,                      // SLPOUT exit sleep mode
    0x13, CMD_DELAY, 10,                      // NORON
    0x3A, 0x01, 0x55,                         // COLMOD set 16-bit pixel format 0x55
    0x36, 0x01, 0b00001000,                   // MADCTL 08 Adafruit
    0xB6, 0x02, 0x0A, 0x82,                   // DISSET5
    0xB2, 0x05, 0x0C, 0x0C, 0x00, 0x33, 0x33, // FRMCTR2 / PORCTL, Frame Rate Control (In Idle mode/ 8-colors)
    0xB7, 0x01, 0x35,                         // VGH / VGL
    0xBB, 0x01, 0x28,                         // VCOM
    0xC0, 0x01, 0x0C,                         // LCM / PWCTR1 power control 1
    0xC2, 0x02, 0x01, 0xFF,                   // VDV PWCTR3 power control 3
    0xC3, 0x01, 0x10,                         // VRH
    0xC4, 0x01, 0x20,                         // VDV
    0xC6, 0x01, 0x0F,                         // FRCTR2
    0xD0, 0x02, 0xA4, 0xA1,                   // PWCTRL1
    0xE0, 0x0E,                               // GMCTRP1 positive gamma correction
    0xD0, 0x00, 0x02, 0x07, 0x0A, 0x28, 0x32, 0x44, 0x42, 0x06, 0x0E, 0x12, 0x14, 0x17, 0xE1,
    0x0E, // GMCTRN1 negative gamma correction
    0xD0, 0x00, 0x02, 0x07, 0x0A, 0x28, 0x31, 0x54, 0x47, 0x0E, 0x1C, 0x17, 0x1B, 0x1E,
    //    0x2A,  0x04,  0x00,  0x00,  0x00,  0x7F,   // set column address, not needed. set by direct API
    //    0x2B,  0x04,  0x00,  0x00,  0x00,  0x9F,   // set page address, not needed. set by direct API
    0x21, CMD_DELAY, 10,  // INVON (21h): Display Inversion On
    0x29, CMD_DELAY, 120, // DISPON display on
};

```
But I don't know what. Is it possible to add the display in a future release? Thanks.

lexus2k, 2022-06-16:
Can you check the latest commit?

GitMoDu, 2023-12-21:
I have tested the current version and it's not fully working yet. Seems like the offset aren't right.
[image]

lexus2k, 2023-12-26:
@GitMoDu Did you try to use `setOffset` API? It is available for ST7789.