lexus2k/lcdgfx #87: ESP IDF v5 compatibility.

ammaree, 2022-06-15:
We have just upgraded our environment to ESP IDF v5 (Master) and converted all components to meet the new requirements.
All applications compile except for a test app that runs on a ESP-WROVER Devkit v4.1 and that utilises the LCDGFX library.

When attempting to compile this app it comes up with an error as below:
`
In file included from /Users/andremaree/Dropbox/devs/ws/z-repo/lcdgfx/src/lcd_hal/avr/../io.h:71,
                 from /Users/andremaree/Dropbox/devs/ws/z-repo/lcdgfx/src/lcd_hal/avr/i2c_embedded.cpp:25:
/Users/andremaree/Dropbox/devs/ws/z-repo/lcdgfx/src/lcd_hal/avr/../esp/esp32_i2c.h:34:10: fatal error: driver/i2c.h: No such file or directory
   34 | #include "driver/i2c.h"
      |          ^~~~~~~~~~~~~~
`
The exact same error occurs in a number of places:
/ws/z-repo/lcdgfx/src/lcd_hal/avr/i2c_embedded.cpp
/ws/z-repo/lcdgfx/src/lcd_hal/arduino/arduino_spi.cpp
/ws/z-repo/lcdgfx/src/lcd_hal/arduino/arduino_wire.cpp
/ws/z-repo/lcdgfx/src/lcd_hal/arduino/platform.cpp
/ws/z-repo/lcdgfx/src/lcd_hal/avr/i2c_twi.cpp
/ws/z-repo/lcdgfx/src/lcd_hal/avr/platform.cpp
/ws/z-repo/lcdgfx/src/lcd_hal/avr/spi_usi.cpp
/ws/z-repo/lcdgfx/src/lcd_hal/esp/esp32_i2c.cpp
/ws/z-repo/lcdgfx/src/lcd_hal/avr/spi_avr.cpp
/ws/z-repo/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp
and many more...

Any suggestions?

lexus2k, 2022-06-15:
Hi, I assume that there are some changed defines with new version ESP IDF v5.
I see that it builds path including AVR directory, which is wrong for ESP32.
I need to double check

lexus2k, 2022-06-16:
Can you try the latest fix I committed?

ammaree, 2022-06-18:
Looks good, building my side...

closed 2022-06-18 by ammaree