lexus2k/lcdgfx #88: ESP32 SPI driver missing when compiling for ESP8266 in Arduino IDE

charel, 2022-07-01:
**Describe the bug**
I'm trying to setup a SH1106 1.3" display using this library. Unfortunately whenever I try to compile it in the IDE, even with the minimal example given in the README.md, I get the error of missing "driver/spi_master.h".

**To Reproduce**
Steps to reproduce the behavior:
1. Installed fresh version of Arduino IDE
2. Added ESP8266 and ESP32 boards to IDE using the official Espressif github links
3. Installed lcdgfx library through IDE
4. Setup NodeMCU 1.0
5. Try to compile example for this library (either demo or just minimal from README.md)

**Expected behavior**
Expected at least the demo to compile.

**Screenshots**
[image]

**Please complete the following information:**
 - library version: 1.1.4
 - LCD display type: knock-off 1.3" SH1106 white OLED
 - OS: macos
 - Platform: ESP8266/NodeMCU
 - IDE: Arduino IDE v. 1.8.19
 
**Additional context**
I tried using different versions of this library and the espressif board files, but I got the same errors regardless.
I also tried removing the ESP32 dependencies, but that just got me into more errors than before.

klsdfsdjfj, 2022-07-15:
any fixes???

lexus2k, 2022-07-17:
It would be nice, if someone has the fix already. If not, then the problem will be fixed, when I have the time.
The library is free.

lexus2k, 2022-07-17:
My compilation passes for ESP8266: 
[image]

charel, 2022-07-19:
Even though I usually compile for a NodeMCU, using the generic ESP8266 gives me the same error as mentioned in my OP.

kileo123, 2022-07-23:
I'm facing the same issue while using this library, and since i dont have ESP32, so what i did was disable the include ESP32 in io.h then "remove" the esp32 files from lcd_hal/esp library.
By remove i mean rename it to something else not recognizable by arduino ide.
it works for me now (NodeMCU and ST7735)

lexus2k, 2022-07-31:
Still the same.

I downloaded the latest board support package for ESP8266, selected NodeMCU 1.0 (12E), and still do not observe any compilation errors.
Maybe there is an issue with very specific configuration of Arduino environment.
Can you send me more details:

The version of Arduino IDE
The version of the board support package you use, how did you install them.
The exact code example, you try to compile

Knabbler, 2022-09-02:
I have the same problem with ESP8266 NodeMCU. 
No problems with ESP32 by the way.
Program to compile is the example "sh1106_demo" as provided.
Attached the copy of the error messages:
[Compiling_Error.txt](https://github.com/lexus2k/lcdgfx/files/9480588/Compiling_Error.txt)

alastaira, 2022-10-15:
I have encountered the same issue while targetting the "LOLIN(WEMOS) D1 R2 & mini" ESP8266-based board. 

As far as I can see, the line #include "driver/spi_master.h" should only be included (in esp32_spi.h) if the following directives are both defined:

```
#if defined(CONFIG_ESP32_SPI_AVAILABLE) && defined(CONFIG_ESP32_SPI_ENABLE)
```

And CONFIG_ESP32_SPI_AVAILABLE should only be set (on line 40 of lcd_hal\esp\io.h) if the following is true:

```
#if !defined(CONFIG_TARGET_PLATFORM_ESP8266) && !defined(CONFIG_IDF_TARGET_ESP8266) && !defined(ESP8266) &&            \
    !defined(__ESP8266__)
```

When compiling for the Wemos board, "ESP8266" is defined (the other three are not), so this test should fail, and CONFIG_ESP32_SPI_AVAILABLE should never be set?

lexus2k, 2022-10-16:
Checked on Arduino IDE 1.8.13, ESP8266 Boards package 3.0.2 `LOLIN(WEMOS) D1 R2 & mini`. 
Compilation of ssd1306_demo passes:

```
Executable segment sizes:
ICACHE : 32768           - flash instruction cache 
IROM   : 242724          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27433   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...) 
DATA   : 1532  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 1004  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 26192 )         - zeroed variables      (global, static) in RAM/HEAP 
Using library lcdgfx at version 1.1.0 in folder: Documents\Arduino\libraries\lcdgfx 
Using library SPI at version 1.0 in folder: \AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\SPI 
Using library Wire at version 1.0 in folder: \AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\Wire 
"\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-size" -A "\\AppData\\Local\\Temp\\arduino_build_520709/ssd1306_demo.ino.elf"
Sketch uses 272693 bytes (26%) of program storage space. Maximum is 1044464 bytes.
Global variables use 28728 bytes (35%) of dynamic memory, leaving 53192 bytes for local variables. Maximum is 81920 bytes.

```

alastaira, 2022-10-17:
Thankyou for your attention :) 
Eurgh, well that is frustrating - after a restart,  double-checking and recompiling the sketch again, now I can't recreate the error! I had upgraded from a previous version of the library - perhaps there was some conflict with previous cached version?

alastaira, 2022-11-18:
Ok, so after incrementally upgrading the library and re-testing, I can clarify that the error occurs in a change introduced in v1.1.2  (in other words, versions<=1.1.1 work fine, versions>=1.1.2 and later fail with the compilation error above.
Looking at the change history around that point, I was struck by the following #define inserted into line 129 of src/lcd_hal/arduino/io.h - surely the CONFIG_ESP32_SPI_AVAILABLE flag should only be set if the target architecture is ESP32?

```
#elif defined(ESP8266) || defined(ESP32) || defined(ESP31B) || defined(ARDUINO_ARCH_SAMD)
...
#define CONFIG_ESP32_SPI_AVAILABLE
```

[[src/lcd_hal/arduino/io.h](https://github.com/lexus2k/lcdgfx/compare/v1.1.1...v1.1.2#diff-6163d3280a84173e4fe2210490325a902b865316c467d7e86bd9a5ddc0f7b796)
](https://github.com/lexus2k/lcdgfx/compare/v1.1.1...v1.1.2#diff-6163d3280a84173e4fe2210490325a902b865316c467d7e86bd9a5ddc0f7b796)

Andr1961, 2023-04-01:
>C:\Users\alast\Documents\Arduino\libraries\lcdgfx\src/lcd_hal/esp/esp32_spi.h:34:10: fatal error: driver/spi_master.h: No such file or directory
>    34 | #include "driver/spi_master.h"

it helped me in UserSettings.h

#ifdef ESP32
/** Define this macro if you need to enable ESP32 I2C module for compilation */
#define CONFIG_ESP32_I2C_ENABLE
/** Define this macro if you need to enable ESP32 SPI module for compilation */
#define CONFIG_ESP32_SPI_ENABLE
#endif

#ifdef ESP8266
/** Define this macro if you need to enable ESP8266 I2C module for compilation */
#define CONFIG_ESP8266_I2C_ENABLE
/** Define this macro if you need to enable ESP8266 SPI module for compilation */
#define CONFIG_ESP8266_SPI_ENABLE
#endif

alastaira, 2023-09-13:
ESP32 and ESP8266 are defined by the Arduino IDE at compilation time based on the target platform selection, right? I'm not sure exactly how, but looking through the boards.txt file for the Lolin Wemos D1 Mini board which I'm targetting, I don't see any reference to ESP32, only ESP8266 (which is correct). So I'm not sure why/where/how ESP32 is getting defined?

lexus2k, 2023-09-17:
@alastaira Yes, the library doesn't define these macro. It only uses ESP32, ESP8266, __XTENSA__ and other macro to understand what is the target system. 
I tried to build the library on my PC for different target platforms and do not observe the issue. However I admit that some users experience build issue. If you can enable detailed compilation logs in your system and provide me the list of gcc command line calls with all arguments passed to compiler, it would be nice

reconbot, 2024-11-30:
I've submitted pull request #131 to move the offending definition out of scope for ESP8266 and ARDUINO_ARCH_SAMD.

closed 2024-12-02 by lexus2k