lexus2k/lcdgfx #90: How to driver 256x64 displays?

sevenberyl, 2022-08-14:
An excellent library! 
Does this library support 256x64 types of displays? 
How? Can you give me a hint? Thank you for your efforts!

lexus2k, 2022-08-19:
Hi.

This library can support any display sizes, however, what is the display controller, used in your display?

sevenberyl, 2022-08-20:
HI,Thanks for the reply！
The driver is SH1122，Interface is SPI or IIC

lexus2k, 2022-08-21:
Can you confirm that ssd1327 demo somehow works for your sh1122 display. If you can see something, that means that sh1122 has similar commands, and it will be easier to add 256x64 support.

sevenberyl, 2022-08-22:
Thanks for the tip! I'll give it a try.

lexus2k, 2022-10-12:
@sevenberyl Any success? If ssd1327 works for you I can add support for sh1122.