lexus2k/lcdgfx #93: Error compiling for board Arduino MKRZERO

stealth-ultd, 2022-09-30:
**Describe the bug**
Compiling with lcdgfx versions 1.1.2 and 1.1.4 terminates with the following error:

"In file included from /Users/xx/Documents/Arduino/libraries/lcdgfx/src/lcd_hal/io.h:55:0,
Multiple libraries were found for "Ethernet.h"
 Used: /Users/xx/Documents/Arduino/libraries/Ethernet
 Not used: /Applications/Arduino.app/Contents/Java/libraries/Ethernet
                 from /Users/xx/Documents/Arduino/libraries/lcdgfx/src/nano_gfx_types.h:31,
                 from /Users/xx/Documents/Arduino/libraries/lcdgfx/src/lcdgfx.h:30,
                 from /Users/xx/Documents/Arduino/TA-Regulator-dev55/OledDisplay.ino:3:
/Users/xx/Documents/Arduino/libraries/lcdgfx/src/lcd_hal/esp/esp32_spi.h:34:10: fatal error: driver/spi_master.h: No such file or directory
 #include "driver/spi_master.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino MKRZERO."

**>> Note: With version lcdgfx 1.1.1 all works OK.**

**To Reproduce**
Steps to reproduce the behavior:
1. update from lcdgfx version 1.1.1 to 1.1.2 or 1.1.4 
2. compile > gives error, compiler exits
3. downgrade to 1.1.1
4. compile > no error, all = OK

**Expected behavior**
No compiling error occurs and sketch should complete compilation without error.

**Please complete the following information:**
 - version > 1.1.1
 - SSD1327 Oled 128 x 128
 - MacOS
 - SAMD21 with Boards Package 1.8.13
 - IDE 1.8.16

**Appreciate this library a lot, and hope also the updated version (1.1.4) can be made to work with MKRZERO!**

lexus2k, 2022-10-02:
Arduino IDE 1.8.13
SAMD21 with Board Package 1.6.12 (the latest)

Everything is compiled successfully  (Windows 11 OS)
Unfortunately, I don't have Mac. I would really appreciate if you find the root cause.

[image]
[image]

Thank you

lexus2k, 2022-10-02:
One more thing. Can you try the latest library code from master branch?
If it works for you, then the issue will be fixed with newer lcdgfx release

stealth-ultd, 2022-10-02:
Thanks for looking into this!

This is what the latest library (Master branch) gives:
"In file included from /Users/xx/Documents/Arduino/libraries/lcdgfx/src/lcd_hal/io.h:55:0,
                 from /Users/xx/Documents/Arduino/libraries/lcdgfx/src/nano_gfx_types.h:31,
                 from /Users/xx/Documents/Arduino/libraries/lcdgfx/src/lcdgfx.h:30,
                 from /Users/xx/Documents/Arduino/TA-Regulator-dev57/OledDisplay.ino:3:
/Users/xx/Documents/Arduino/libraries/lcdgfx/src/lcd_hal/esp/esp32_spi.h:34:10: fatal error: driver/spi_master.h: No such file or directory
 #include "driver/spi_master.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Multiple libraries were found for "Ethernet.h"
 Used: /Users/xx/Documents/Arduino/libraries/Ethernet
 Not used: /Applications/Arduino.app/Contents/Java/libraries/Ethernet
exit status 1
Error compiling for board Arduino MKRZERO."

BTW: The latest board package for the SAMD board is 1.8.13: https://github.com/arduino/ArduinoCore-samd/releases
In your screenshot, it looks like you compile with the SAM package (for Arduino Due), not the SAMD one (for MKRZERO), if I'm correct?

lexus2k, 2022-10-09:
```.cpp
#if defined(CONFIG_ESP32_SPI_AVAILABLE) && defined(CONFIG_ESP32_SPI_ENABLE)
```

For some reaon, this condition is true in your case. Those constants are defined in src/lcd_hal/esp/io.h, which is used only when you have ESP32 boards: 
```.cpp
#elif defined(__XTENSA__) && !defined(ARDUINO)
#include "esp/io.h"
```

You can try to debug in that direction.

stealth-ultd, 2023-01-27:
I finally have found a workaround:
comment out in lcdgfx/src/lcd-hal/UserSettings.h
//#define CONFIG_ESP32_I2C_ENABLE
//#define CONFIG_ESP32_SPI_ENABLE
It does not solve the origin of this problem, but at least I can now compile versions newer than 1.1.1.

closed 2024-12-02 by lexus2k

stealth-ultd, 2024-12-02:
Thanks for getting back to this!

It has resolved the ESP32 related compilation error, but not yet the SPI related error:

"Compilation error: exit status 1

/Users/XXX/Documents/Arduino/libraries/lcdgfx/src/lcd_hal/esp/esp32_spi.h:34:10: fatal error: driver/spi_master.h: No such file or directory
 #include "driver/spi_master.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1"

–> I can fix it manually by commening out in lcdgfx/src/lcd-hal/UserSettings.h: //#define CONFIG_ESP32_SPI_ENABLE

Hope this helps!

lexus2k, 2024-12-02:
@stealth-ultd 

The main issue here is that they (Espressif) completely changed i2c API interface in ESP IDF 5.X and higher, making it totally incompatible.
The latest supported ESP IDF sdk is 4.X.  It is possible to support ESP IDF 5.X, but it requires time.