lexus2k/lcdgfx #95: HSPI_HOST and VSPI_HOST deprecated for ESP32-S3 in ESP-IDF

vintzvintz, 2022-11-27:
**Context**

Hobby project with plain ESP-IDF, **not Arduino Core**, on a  EPS32-S3 devkit board.
Using platformio/vscode on linux
Added a "components" subdir in projet root and cloned lcdgfx repo into it.
I2C SSD1306 oled display, so SPI support is not needed in my case.

Build fails on file esp32_spi.cpp with a bunch of errors like **error: 'VSPI_HOST' was not declared in this scope**
( see build logs below)

**Analysis** 

HSPI_HOST and VSPI_HOST are deprecated and not defined anymore in latest versions of IDF and/or on newer ESP32 hardwares, as described at the end of spi_types.h IDF header file.

**Workaround**

I try and failed to completely disable SPI support by commenting out some lines in src/lcd_hal/UserSettings.h. 
It looks like at least 1 SPI implementation is required, and i do not have Arduino Wire in my project.

As a workaround, I added aliases in src/lcd_hal/esp/esp32_spi.cpp just after original #includes.

#ifdef CONFIG_IDF_TARGET_ESP32S3
    #define SPI_HOST    SPI1_HOST
    #define HSPI_HOST   SPI2_HOST
    #define VSPI_HOST   SPI3_HOST
#endif

**Conclusion**
I hope @lexus2k or someone more skilled/experienced than me would provide a cleaner fix for these deprecated VSPI_HOST and HSPI_HOST symbols on latest ESP32 family members (after ESP32 S2)

**Please complete the following information:**
 - library version :  commit 058b564a
 - LCD display type : SSD1306 128x64 I2C
 - OS : linux
 - Platform : esp32-S3
 - IDE : Vscode/platformio

**Additional context**
Processing esp32-s3-devkitc-1 (platform: espressif32; board: esp32-s3-devkitc-1; framework: espidf)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (5.2.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40201.0 (4.2.1) 
 - tool-ninja @ 1.7.1 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/esp32-s3-devkitc-1/components/lcdgfx/src/lcd_hal/esp/esp32_spi.o
components/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp: In member function 'void EspSpi::begin()':
components/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp:96:34: error: 'VSPI_HOST' was not declared in this scope
     spi_bus_initialize(m_busId ? VSPI_HOST : HSPI_HOST, &buscfg, 0); // 0 -no dma
                                  ^~~~~~~~~
components/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp:96:34: note: suggested alternative: 'SPI3_HOST'
     spi_bus_initialize(m_busId ? VSPI_HOST : HSPI_HOST, &buscfg, 0); // 0 -no dma
                                  ^~~~~~~~~
                                  SPI3_HOST
components/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp:96:46: error: 'HSPI_HOST' was not declared in this scope
     spi_bus_initialize(m_busId ? VSPI_HOST : HSPI_HOST, &buscfg, 0); // 0 -no dma
                                              ^~~~~~~~~
components/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp:96:46: note: suggested alternative: 'SPI3_HOST'
     spi_bus_initialize(m_busId ? VSPI_HOST : HSPI_HOST, &buscfg, 0); // 0 -no dma
                                              ^~~~~~~~~
                                              SPI3_HOST
components/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp: In member function 'void EspSpi::end()':
components/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp:113:28: error: 'VSPI_HOST' was not declared in this scope
     spi_bus_free(m_busId ? VSPI_HOST : HSPI_HOST);
                            ^~~~~~~~~
components/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp:113:28: note: suggested alternative: 'SPI3_HOST'
     spi_bus_free(m_busId ? VSPI_HOST : HSPI_HOST);
                            ^~~~~~~~~
                            SPI3_HOST
components/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp:113:40: error: 'HSPI_HOST' was not declared in this scope
     spi_bus_free(m_busId ? VSPI_HOST : HSPI_HOST);
                                        ^~~~~~~~~
components/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp:113:40: note: suggested alternative: 'SPI3_HOST'
     spi_bus_free(m_busId ? VSPI_HOST : HSPI_HOST);
                                        ^~~~~~~~~
                                        SPI3_HOST
components/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp: In member function 'void EspSpi::start()':
components/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp:139:38: error: 'VSPI_HOST' was not declared in this scope
         spi_bus_add_device(m_busId ? VSPI_HOST : HSPI_HOST, &devcfg, &m_spi);
                                      ^~~~~~~~~
components/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp:139:38: note: suggested alternative: 'SPI3_HOST'
         spi_bus_add_device(m_busId ? VSPI_HOST : HSPI_HOST, &devcfg, &m_spi);
                                      ^~~~~~~~~
                                      SPI3_HOST
components/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp:139:50: error: 'HSPI_HOST' was not declared in this scope
         spi_bus_add_device(m_busId ? VSPI_HOST : HSPI_HOST, &devcfg, &m_spi);
                                                  ^~~~~~~~~
components/lcdgfx/src/lcd_hal/esp/esp32_spi.cpp:139:50: note: suggested alternative: 'SPI3_HOST'
         spi_bus_add_device(m_busId ? VSPI_HOST : HSPI_HOST, &devcfg, &m_spi);
                                                  ^~~~~~~~~
                                                  SPI3_HOST
*** [.pio/build/esp32-s3-devkitc-1/components/lcdgfx/src/lcd_hal/esp/esp32_spi.o] Error 1

lexus2k, 2022-11-29:
Hello @vintzvintz 

Do you have any fix to share?

lexus2k, 2023-02-12:
It should be fixed with the recent commits.
Feel free to reopen it if the issue still exists

closed 2023-02-12 by lexus2k