lexus2k/lcdgfx #96: disabling all "SPI" or "I2C" macros in "UserSettings.h" causes a compilation error.

LuKkas2101, 2022-11-27:
to reproduce the error is easy just open an example of i2c. I used this "menu_demo.ino".
comment all settings related to SPI.

enabling "#define CONFIG_ARDUINO_SPI_ENABLE" or "#define CONFIG_AVR_SPI_ENABLE", resolves the error.

the problem seems to be caused by classes referencing base classes.

**Expected behavior**

I expected that when using i2c classes and removing the spi macros, only the i2c classes would be compiled. as described in UserSettings.h

 - library version ----------- 1.1.4
 - LCD display type---------SSD1306_128x64_I2C
 - OS ----------------------- windows
 - Platform ----------------- arduino nano old bootloader Atmega328p
 - IDE if using some --------platformIo VScode

lexus2k, 2022-11-29:
@LuKkas2101 

Thank you for reporting the issue. If you have any fix ready for pull request, let me know

lexus2k, 2022-11-29:
```.shell
In file included from c:\Users\Alex\Documents\Arduino\libraries\lcdgfx\src/nano_gfx_types.h:31:0,
                 from c:\Users\Alex\Documents\Arduino\libraries\lcdgfx\src/lcdgfx.h:30,
                 from C:\Users\Alex\AppData\Local\Temp\.arduinoIDE-unsaved20221029-28016-22wp1n.z2kf5\ssd1306_demo\ssd1306_demo.ino:39:
c:\Users\Alex\Documents\Arduino\libraries\lcdgfx\src/lcd_hal/io.h:625:2: error: #error "Platform not supported"
 #error "Platform not supported"
  ^~~~~
```

lexus2k, 2022-11-29:
@LuKkas2101 Why are you trying to disable all SPI interfaces? This will not reduce memory usage