lexus2k/lcdgfx #100: I2C frequency gets narrowed

ggatis, 2023-02-22:
Display works with default constructor.
Error using custom I2C frequency: DisplaySH1107_64x128_I2C display( -1, { -1, 0x3C -1, -1, 1000000 } );

lcdgfx_sh1107_demo:46:68: error: narrowing conversion of '1000000' from 'long int' to 'int8_t {aka signed char}' inside { } [-Wnarrowing]
 DisplaySH1107_64x128_I2C display( -1, { -1, 0x3C -1, -1, 1000000 } );
                                                                    ^
Using library lcdgfx at version 1.1.4 in folder: ...\Arduino\libraries\lcdgfx 
Using library SPI at version 1.0 in folder: ...\ArduinoData\packages\lgt8fx\hardware\avr\2.0.0\libraries\SPI 
Using library Wire at version 1.0 in folder: ...\ArduinoData\packages\lgt8fx\hardware\avr\2.0.0\libraries\Wire 
exit status 1
narrowing conversion of '1000000' from 'long int' to 'int8_t {aka signed char}' inside { } [-Wnarrowing]

I suspect
  DisplaySH1107_64x128_I2C display( -1, { -1, 0x3C -1, -1, 1000000 } );
should set I2C frequency to 1MHz.
Changing in file lcd_sh1107.h "SPlatformI2cConfig{config.busId, static_cast<uint8_t>(config.addr ?: 0x3C), config.scl, config.sda, config.frequency ?: 400000})" to "SPlatformI2cConfig{config.busId, static_cast<uint8_t>(config.addr ?: 0x3C), config.scl, config.sda, config.frequency ?: 1000000})" does not give any effect.
Changing in file arduino_wire.cpp "Wire.setClock( 400000 );" to "Wire.setClock( 1000000 );" is efficient.
The effect was observed commenting all lcd_delay functions in demo file, initialising Serial and printing out demo execution time.
Changes in demo file:
----------------------
...
void setup() {

    Serial.begin(57600);
...

unsigned long StartMillis;
unsigned long CurrentMillis;

void loop()
{
    //lcd_delay(1000);
    switch (menu.selection())
    {
        case 0:
            StartMillis  = millis();
            bitmapDemo();
            break;
...
        case 4:
            drawLinesDemo();
            CurrentMillis = millis();
            Serial.print("Test duration, ms:");
            Serial.println( CurrentMillis - StartMillis );
            break;

...
----------------------
The execution time changed from default 1503 ms to 808 ms.
At I2C frequency 2 MHz it decreased to 576 ms.

 - library version: 1.1.4
 - LCD display type SH1107 64x128 I2C
 - OS: Windows 10
 - Platform: LGT8F328P
 - IDE: Arduino 1.8.16 from MS store

lexus2k, 2023-02-23:
Hi @ggatis 

the frequency is not a last parameter in SPlatformI2cConfig. Probably you should you something like this:
DisplaySH1107_64x128_I2C display( -1, { -1, 0x3C, 1000000, -1, -1, } );

ggatis, 2023-02-23:
Hi @lexus2k!
I took the syntax from the demo where it says
DisplaySH1107_128x64_I2C display(-1); // or (-1,{busId, addr, scl, sda, frequency})...
I think it is the correct sequence since the second arg should go as config for structure SPlatformI2cConfig that in short is defined as
typedef struct {
        int8_t busId;
        uint8_t addr;
        int8_t scl;
        int8_t sda;
        uint32_t frequency;
} SPlatformI2cConfig;

And "DisplaySH1107_64x128_I2C display( -1, { -1, 0x3C, 1000000, -1, -1, } );" does not work anyway since the prob is in narrowing, not in the order:

lcdgfx_sh1107_demo:47:70: error: narrowing conversion of '1000000' from 'long int' to 'int8_t {aka signed char}' inside { } [-Wnarrowing]
 DisplaySH1107_64x128_I2C display( -1, { -1, 0x3C, 1000000, -1, -1, } );
                                                                      ^
lcdgfx_sh1107_demo:47:70: error: narrowing conversion of '-1' from 'int' to 'uint32_t {aka long unsigned int}' inside { } [-Wnarrowing]
Using library lcdgfx at version 1.1.4 in folder: ...\Arduino\libraries\lcdgfx 
Using library SPI at version 1.0 in folder: ...\ArduinoData\packages\lgt8fx\hardware\avr\2.0.0\libraries\SPI 
Using library Wire at version 1.0 in folder: ...\ArduinoData\packages\lgt8fx\hardware\avr\2.0.0\libraries\Wire 
exit status 1
narrowing conversion of '1000000' from 'long int' to 'int8_t {aka signed char}' inside { } [-Wnarrowing]

Kind regards ~

lexus2k, 2023-03-04:
I do not observe any narrowing error. 
Could you please specify platform you're using for the compilation?