lexus2k/lcdgfx #102: How to drive a 3-wire SPI display?

gaaf, 2023-05-02:
Hi,

I'm trying to drive an oled [display](https://www.aliexpress.com/item/4001035404634.html) in 3-wire SPI mode (no D/C line).

I initialize lcdgfx with DC pin set to -1. Display stays black no matter what I try.

```
#define VSPI_MOSI_PIN    2
#define VSPI_CLK_PIN    15
#define VSPI_CS_PIN     12
#define VSPI_DC_PIN     -1
#define VSPI_RST_PIN    -1

DisplaySSD1351_128x128x16_SPI display(VSPI_RST_PIN,{-1, VSPI_CS_PIN, VSPI_DC_PIN, 0, VSPI_CLK_PIN, VSPI_MOSI_PIN});
```

This oled is the only SPI device I have, so I have no way to verify if the problem is in the oled, lcdgfx or the platform.
At least with 4-wire SPI the display is working fine.

Is 3-wire SPI supported by llcdgfx

(on the oled pcb, I've set the BS jumper to 1).

lexus2k, 2023-05-03:
Hello Alex
Only 4-wire SPI interface is supported by the library.

Best regards

lexus2k, 2026-04-03:
3-wire SPI (without DC pin) requires a different protocol where the D/C bit is sent as the 9th bit. This is not currently supported and would require significant HAL changes.