lexus2k/lcdgfx #104: ESP32 doesn't find library

gregfox151, 2023-06-24:
Ask any question, you have, regarding the library
ON complying an ESP32 LOLIN32, I get this message:
FQBN: esp32:esp32:lolin32

In file included from C:\Users\greg\Documents\Arduino\OLED\OLED_Test-OK\lcdgfx.h:30:0,
                 from C:\Users\greg\Documents\Arduino\OLED\OLED_Test-OK\OLED_Test-OK.ino:3:
C:\Users\Documents\Arduino\OLED\OLED_Test-OK\nano_gfx_types.h:31:24: fatal error: lcd_hal/io.h: No such file or directory
compilation terminated.
Alternatives for lcd_hal/io.h: []
ResolveLibrary(lcd_hal/io.h)
  -> candidates: []

exit status 1

Compilation error: lcd_hal/io.h: No such file or directory
______________
I have included;
#include "lcdgfx.h"
#include "lcdgfx_gui.h"
#include "esp_system.h"
But I still get the error, what can be wrong

lexus2k, 2023-06-29:
Hi. If you're using Arduino IDE please install the library like all other regular Arduino libraries to libraries subfolder.

lexus2k, 2026-04-03:
This is an Arduino IDE library path issue, not a library bug. Ensure lcdgfx is installed via Library Manager or placed correctly in your Arduino/libraries/ directory.

closed 2026-04-03 by lexus2k