lexus2k/lcdgfx #106: LCDGFX doesn't appear pixel aligned

schwos, 2023-08-03:
So we have been positioning text and graphics on our SSD1306 OLED with the library. When we expect to move text to say 45, 32 it doesn't appear that it will go through but formats in general the general spot. When you attempt to be precise it can become challenging. 

As well when putting BitXMP on the screen and then a box around it it appears that there is an overlap. Anyway to turn on transparent background so it doesn't turn a pixel off causing flickering as it repaints?

lexus2k, 2023-08-04:
@schwos For this case you need to use NanoEngine to avoid flickering when drawing. Please check the examples section. Snowflakes is a good one.

schwos, 2023-08-08:
I have successfully deployed the nano engine for the project. The one challenge I am working to develop a low ram solution is right justifying text in specific graphic boxes created on the OLED. I was reviewing the source and see a gettextsize within font.cop which uses the fonts width + spacing determine total width and fonts height. 

Is there an easy way to get this information so one could calculate position? It could be a lot less memory then doing char or string adaptations to add space at the beginning of the text to output. 

Appreciate suggestions..

schwos, 2023-08-08:
I figured it out..  engine.getCanvas().getFont().getTextSize(buff)

lexus2k, 2023-08-08:
Since you are working on low ram solution, any attempt to avoid flickering will require a double buffering. The NanoEngine allows to reduce ram usage. but it still consumes a ram. 
The problem of many i2c, 3-wire or 4-wire spi displays is that the interface doesn't allow reading of pixels from GDRAM. So, with direct API when you place text over some picture, there is no way to read old pixels do OR operation and write the new set of pixels back. So, this is a challenge.
Some color displays have a hack to read pixels back, but that's a hack and not normal SPI usage.

schwos, 2023-08-09:
Yep so far RAM need by nano engine, fonts and functions fit with Arduino solution fine. With our ESP32 solution doors open a bit not as restrictive. Thanks