lexus2k/lcdgfx #107: Add ARC to Circle

schwos, 2023-08-03:
This isn't related to any known problem with the 
Circle function just like to port some code from u2g2 to this library. Within u2g2 there is the ability to create an arc using the circle function. 

Basically, it could be added to the Circle function as an ending variable that would be optional. 

This would permit one to create a half circle from the bottom of the screen arching could be used for various tools like simulated analog volt meter or vu meter.

Let me know if there is more information requested.

lexus2k, 2023-08-08:
Please check the mainline, if this is what you were writing about.

schwos, 2023-08-09:
I see we can draw a circle via a drawCircle(lcdint_t xc, lcdint_t yc, lcdint_t r, uint8_t options) but would be beneficial if we could as to only draw saw a 1/4 of it. like top right, top left now you have a half circle from bottom right of screen arc'd to left of screen.

closed 2024-02-12 by lexus2k