lexus2k/lcdgfx #109: Integration of LCDGFX

schwos, 2023-08-19:
I have started the attempted integration of LCDGFX with Nanoengine enabled for the ArduinoMenu Library. I have to initialize the menu and pass the display and nanoengine; NanoEngine1<DisplaySSD1306_128x64_I2C> engine(display); 

I am attempting to figure out how I can pass the nanengine and display objects to ArduinoMenu without throwing a compiler error. 

I have tried: 

> MENU_OUTPUTS(out, MAX_DEPTH, NanoEngine1Out(engine, colors, fontX, fontY, offsetX, offsetY, {0, 0, U8_Width / fontX, U8_Height / fontY})
>              ,NONE);

> MENU_OUTPUTS(out, MAX_DEPTH, NanoEngine1Out(engine(display), colors, fontX, fontY, offsetX, offsetY, {0, 0, U8_Width / fontX, U8_Height / fontY})
>              ,NONE);

Suggestions would be appreciated..

------------------------------------------------------------------------------------------------------------

Below is the draft lcdgfxOut.h library to be inserted into ArduinoMenu.. 

`#ifndef RSITE_ARDUINOP_MENU_LCDGFX
#define RSITE_ARDUINOP_MENU_LCDGFX
//#include <U8g2lib.h>
#include "lcdgfx.h"
#include "lcdgfx_gui.h"
//DisplaySSD1306_128x64_I2C display(-1); // or (-1,{busId, addr, scl, sda, frequency}). This line is suitable for most platforms by default
//NanoEngine1<DisplaySSD1306_128x64_I2C> engine(display);

#include "menuDefs.h"

namespace Menu {

	class NanoEngine1Out:public gfxOut {
		public:
      int8_t offsetX=0;
      int8_t offsetY=0;
			NanoEngine1<DisplaySSD1306_128x64_I2C>& gfx;
			//NanoEngine1& gfx;
			const colorDef<uint8_t> (&colors)[nColors];
			NanoEngine1Out(
				NanoEngine1<DisplaySSD1306_128x64_I2C>& gfx,
				//NanoEngine1& gfx,
				const colorDef<uint8_t> (&c)[nColors],
				idx_t* t,
				panelsList &p,
				idx_t resX=6,
				idx_t resY=9,
				idx_t offsetX=0,
				idx_t offsetY=0,
        int fontMarginY=1
			) :gfxOut(resX,resY,t,p,(styles)(menuOut::redraw|menuOut::rasterDraw)),gfx(gfx),colors(c) {
					//gfx.getCanvas().setOffset();
					//gfx.setFontPosBottom(); // U8Glib font positioning
					this->offsetX=offsetX;
					this->offsetY=offsetY;
          this->fontMarginY=fontMarginY;
			}

			NanoEngine1Out(
				NanoEngine1<DisplaySSD1306_128x64_I2C>& gfx,
				const colorDef<uint8_t> (&c)[nColors],
				idx_t* t,
				panelsList &p,
				idx_t resX,
				idx_t resY,
				idx_t offsetX,
				idx_t offsetY,
				int fontMarginX,
				int fontMarginY
			) :gfxOut(resX,resY,t,p,(styles)(menuOut::redraw|menuOut::rasterDraw)),gfx(gfx),colors(c) {
					//gfx.setFontPosBottom(); // U8Glib font positioning
					this->offsetX=offsetX;
					this->offsetY=offsetY;
					this->fontMarginX=fontMarginX;
					this->fontMarginY=fontMarginY;
			}

			size_t write(uint8_t ch) override {return gfx.getCanvas().printChar(ch);}

			inline uint8_t getColor(colorDefs color=bgColor,bool selected=false,status stat=enabledStatus,bool edit=false) const {
				return memByte(&(stat==enabledStatus?colors[color].enabled[selected+edit]:colors[color].disabled[selected]));
			}

			void setColor(colorDefs c,bool selected=false,status s=enabledStatus,bool edit=false) override {
				gfx.getCanvas().setColor(getColor(c,selected,s,edit));
			}

			void clearLine(idx_t ln,idx_t panelNr=0,colorDefs color=bgColor,bool selected=false,status stat=enabledStatus,bool edit=false) override {
				const panel p=panels[panelNr];
				setColor(color,selected,stat,edit);
				gfx.getCanvas().fillRect(p.x*resX + offsetX /*- 1*/,(p.y+ln)*resY + offsetY /*- fontMarginY*/,maxX()*resX /*+ fontMarginY*/ /*+ fontMarginY*/,resY /*+ fontMarginY*/ /*+ fontMarginY*/);
				//setCursor(0,ln);
			}
			void clear() override {
				setCursor(0,0);
				setColor(fgColor);
				panels.reset();
			}
			void clear(idx_t panelNr) override {
				const panel p=panels[panelNr];
				setColor(bgColor,false,enabledStatus,false);
				gfx.getCanvas().fillRect(p.x*resX + offsetX,p.y*resY + offsetY,p.w*resX,p.h*resY);
				//clear();
				panels.nodes[panelNr]=NULL;
			}
			void box(idx_t panelNr,idx_t x,idx_t y,idx_t w=1,idx_t h=1,colorDefs c=bgColor,bool selected=false,status stat=enabledStatus,bool edit=false) override {
				const panel p=panels[panelNr];
				gfx.getCanvas().drawRect((p.x+x)*resX,(p.y+y)*resY,w*resX,h*resY/*+(fontMarginY<<1)*/);
			}

			void rect(idx_t panelNr,idx_t x,idx_t y,idx_t w=1,idx_t h=1,colorDefs c=bgColor,bool selected=false,status stat=enabledStatus,bool edit=false) override {
				const panel p=panels[panelNr];
				gfx.getCanvas().fillRect((p.x+x)*resX,(p.y+y)*resY,w*resX,h*resY/*+(fontMarginY<<1)*/);
			}

			void setCursor(idx_t x,idx_t y,idx_t panelNr=0) override {
				// _trace(Serial<<"setCursor"<<endl);
				const panel p=panels[panelNr];
				gfx.getDisplay().setTextCursor((p.x+x)*resX+fontMarginX + offsetX,(p.y+y+1)*resY-fontMarginY + offsetY);
				//gfx.getCanvas().printFixedPgm((p.x+x)*resX+fontMarginX + offsetX,(p.y+y+1)*resY-fontMarginY + offsetY,"");
				//gfx. = (p.x+x)*resX+fontMarginX + offsetX;
				//gfx.ty = (p.y+y+1)*resY-fontMarginY + offsetY;
			}

			idx_t startCursor(navRoot& root,idx_t x,idx_t y,bool charEdit,idx_t panelNr) override {
			  if (charEdit) {
			    // rect(panelNr,  x-1,  y, 1, 1, bgColor, false, enabledStatus, false);
					const panel p=panels[panelNr];
					gfx.getCanvas().drawRect((p.x+x)*resX + offsetX +fontMarginX,(p.y+y)*resY+ offsetY /*-fontMarginY*/,resX,resY);
			    setColor(fgColor,false,enabledStatus,false);
			  }/* else
			    box(panelNr,  x,  y, 1, 1, bgColor, false, enabledStatus, false);*/
			  return 0;
			}

			// idx_t endCursor(navRoot& root,idx_t x,idx_t y,bool charEdit,idx_t panelNr) override {
			//   setColor(fgColor,true,enabledStatus,true);return 0;
			// }

      idx_t editCursor(navRoot& root,idx_t x,idx_t y,bool editing,bool charEdit,idx_t panelNr=0) override {
			  if (editing) {
					_trace(Serial<<"editCursor"<<endl);
					// box(panelNr,x-1,y);
					const panel p=panels[panelNr];
					gfx.getCanvas().drawRect((x+p.x-1)*resX + offsetX + fontMarginX-1,(p.y+y)*resY + offsetY /*- fontMarginY+2*/,resX+1 ,resY);
				}
			  return 0;
			}

			void drawCursor(idx_t ln,bool selected,status stat,bool edit=false,idx_t panelNr=0) override {
				// _trace(Serial<<"drawCursor"<<endl);
				const panel p=panels[panelNr];
				// gfxOut::drawCursor(ln,selected,stat);
				setColor(cursorColor,selected,stat);
				gfx.getCanvas().drawRect(p.x*resX + offsetX /*+ fontMarginX*/,(p.y+ln)*resY + offsetY /*+ fontMarginY*/,maxX()*resX ,resY);
			}
			idx_t printRaw(const char* at,idx_t len) override {
				trace(Serial<<"NanoEngine1Out::printRaw"<<endl);
				trace(Serial<<"["<<at<<"]");
				return print((__FlashStringHelper*)at);
			  // const char* p=at;
			  // uint8_t ch;
			  // for(int n=0;(ch=memByte(at++))&&(len==0||n<len);n++) {
				// 	trace(Serial<<"["<<ch<<"]");
			  //   write(ch);
			  // }
			  // return at-p-1;
			}

	};
}
#endif //RSITE_ARDUINOP_MENU_U8G2
`

schwos, 2023-08-19:
Within the ArduinoMenu I was successful in getting the MACRO that was develop to initialize to function and when the compiler executes this is the output:

I was successful in getting the MENU_OUTPUTS to generate the initialize:

`Menu::idx_t outTops73[2];  panel _panels_outPanels73[] ={{0, 0, 128 / 6, 64 / 10}};  Menu::navNode* _nodes_outPanels73[sizeof(_panels_outPanels73)/sizeof(panel)];  Menu::panelsList outPanels73(_panels_outPanels73,_nodes_outPanels73,sizeof(_panels_outPanels73)/sizeof(panel));; Menu::NanoEngine1Out out73(engine,colors,outTops73,outPanels73,6,10,0,3);   Menu::menuOut*  out_outPtrs[] ={ &out73,  }; Menu::outputsList out(out_outPtrs,sizeof(out_outPtrs)/sizeof(Menu::menuOut*));`

Though at moment I have LCDGFX complaining..

> stl_vector.h: In member function 'std::vector<_Tp, _Alloc>::size_type std::vector<_Tp, _Alloc>::_M_check_len(std::vector<_Tp, _Alloc>::size_type, const char*) const':
> .pio/libdeps/PVMII_Display_ESP32/lcdgfx/src/nano_gfx_types.h:41:19: error: expected unqualified-id before '(' token
>  #define max(a, b) ((a) > (b) ? (a) : (b))

lexus2k, 2023-08-19:
Hello. Do you have any simple project to build with Arduino IDE giving this error?
I can advise you to replace the name of macro (min, max) to lcd_gfx_min, lcd_gfx_max everywhere in lcdfgx library. If this helps, I will update the library

schwos, 2023-08-19:
We have to add to macros the lcdgfx macro i'll provide that shortly. Show that it will generate the appropriate code upon compile. 

Why are we changing the other macro just curious? 

I'm at moment building out the sample code with ArduinoMenu integrated with LCDGFX so we can just use it outside of my other project. I should have this done late this afternoon. 

I use Visual Code with PlatformIO that okay or you want in Arduino IDE? I'll have to reinstall Arduino IDE if so.

schwos, 2023-08-20:
I was successful in getting the two to compile, it required modification to the ArduinoMenus macro.h file. After careful study I was able to find the items it required. Now compile is one thing getting it to produce the menu through the driver is another. I will work this a get a development sample uploaded shortly so either share the outcome or determine what i may be missing.More to come

lexus2k, 2026-04-03:
ArduinoMenu integration is outside the scope of this library. You would need to implement a custom renderer that calls lcdgfx drawing functions.