lexus2k/lcdgfx #116: Add support for Raspberry Pi Pico - Pull Request

simon-77, 2024-04-29:
The library lcdgfx has a long list of supported platforms, but it still lacks support for the *Raspberry Pi Pico* micro controller using the pico C/C++ SDK.

I have implemented some of the `lcd_hal` functions using the pico-sdk, especially for the SPI implementation and also verified this implementation using hardware.

## Implementation
- The hardware implementation is placed in the directory `lcdgfx/src/lcd_hal/pico/`
- To enable this implementation, the define `PICO_BOARD` is used, which is defined when compiling for the Raspberry Pi Pico and can be used to distinguish between the two boards `pico` and `pico_w` (no difference between these boards is made here)
- The i2c hal is not implemented, as I do not have hardware to verify this. Instead I added empty dummy functions (to be able to compile this project) and added a compiler warning `#warning "I2C is not implemented for Pico board - this is a dummy"`
- The lcdgfx library uses the `LCD_PROGMEM` macro for data stored in flash. (e.g. `const PROGMEM uint8_t heartImage[8]`). The pico-sdk uses the `__in_flash()` macro after type definition like: `const uint8_t __in_flash() heartImage[8]`.
It is therefore unfortunately not possible to use the `LCD_PROGMEM` macro only using precompiler directives. I have defined the `LCD_PROGMEM` "blank" to be able to compile this library.

## Using this library
As the Raspberry Pi Pico suite also uses CMake (and ninja) for the projects, it is fairly straight forward to "include" the source of this library (at least once I figured it out correctly).

I have provided an example with 4 additional statements in the CMakeList.txt of the Raspberry Pi Pico project to include the lcdgfx CMake folder, as well as to compile and link it correctly. Using the lcdgfx library does work the same as on the Arduino platform.

With this approach none additional compile or link process is required, as it is seemingly implemented into the CMake build process of the Raspberry Pi Pico prject.

Best regards
Simon

simon-77, 2024-04-29:
One workaround I forgot to mention regarding the SPI hardware:

The Rapsberry Pi Pico has 2 SPI hardware peripherals (`spi0` and `spi1`) which can each be routed to multiple different pins.
The pin routing is implemented in the pico-hal and can be achieved by providing the correct pin numbers to the constructor of lcdgfx. As there are no additional  parameters in the constructor, I used a define to distinguish between these 2 SPI peripherals

I have therefore introduced the following snippet of the `pico_spi.h` file:
```
#if defined(PICO_USE_SPI1)
#define PICO_SPI spi1
#else
#define PICO_SPI spi0
#endif
```

Maybe not the most elegant solution ... but I did not know how to do it otherwise.

lexus2k, 2024-05-05:
Hi @simon-77 
You solution is merged to the project.
It passed automatic checked and conform code style.
Thank you

gaby64, 2024-09-09:
Hopefully someone can tackle the i2c implementation, im not having very good success in my attempt.
https://github.com/gaby64/lcdgfx
I managed to initialize an ssh1106 and display some text but most things dont work, like clear or fill.
I used code from https://github.com/martinkooij/pi-pico-ss-oled/ to try and make it work

On my next attempt I will try and use the arduino-pico wire code from:
https://github.com/earlephilhower/arduino-pico/tree/master/libraries/Wire/src

simon-77, 2024-09-09:
Hi @gaby64 

It's great that you want to add I2C support for the Pi Pico and I would like to help you by pointing your attention to the right files.

1. to support multiple platforms (Arduino, Pi Pico SDK, ...) a _Hardware Abstraction Layer_ (HAL) is used -> the lcdgfx library defines a class structure, but for each platform this class is implemented differently. The HALs are located here: https://github.com/lexus2k/lcdgfx/tree/master/src/lcd_hal/
2. In order for the project to compile, I already had to declare a I2C HAL class for the pico - I only let the definition of the class functions empty.
3. The Pico C SDK for the I2C hardware is documented here: https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_hardware_i2c and is in general somewhat similar to the Arduino Framework, as both are used for controlling the peripheral of a microcontroller.

So have a look at the I2C HAL implementation for Arduino: https://github.com/lexus2k/lcdgfx/tree/master/src/lcd_hal/arduino
(called "arduino_wire") and try to create a HAL implementation for the Pi Pico.
The header file for the Pi Pico (`pico_i2c.h`) should already contain everything necessary and you only have to "fill in" the empty class functions in the source file: https://github.com/lexus2k/lcdgfx/blob/master/src/lcd_hal/pico/pico_i2c.cpp

---
2 more remarks:

1. Of course you are free of using whatever i2c implementation you like, but I would encourage you to use the Pico C SDK directly (I have linked above) instead of another library, because the whole lcdgfx library would then become dependent on the other i2c library you have linked in your post.

2. The changes you made so far don't look too sensible while glancing over it, I would suggest you start from the latest lcdgfx commit again and start only modifying the files in the `lcd_hal` folder, focusing on the `pico_i2c.cpp` source file.

If you have more questions or stumble across compile errors, feel free to ask again.
Cheers Simon

gaby64, 2024-09-09:
I managed to get it to work, I placed all the code inside pico_i2c.cpp so there is no dependence on another library.

simon-77, 2024-09-09:
Excellent, well done :+1: :smile: 

If you have the changes in a forked repo on github (like you had with your first attempt), you can issue a pull-request to contribute this work to this project.

Cheers, Simon