lexus2k/lcdgfx #119: Build Failing on ESP32-C3 - No SPI3_HOST

falo2k, 2024-07-02:
**Describe the bug**
The build fails for the ESP32-C3 due to SPI3_HOST not being defined.  Error hits in the macros at the top of esp32_spi.cpp where it tries to `#define VSPI_HOST SPI3_HOST`

**To Reproduce**
Steps to reproduce the behavior:
Build a project against `ESP32C3 Dev Module`

```
esp32_spi.cpp: In member function void EspSpi::begin()
 
esp32_spi.cpp: 37:25: error: 'SPI3_HOST' was not declared in this scope; did you mean 'SPI2_HOST'?
   37 |     #define VSPI_HOST   SPI3_HOST
   |                         ^~~~~~~~~
esp32_spi.cpp:103: note  in expansion of macro VSPI_HOST
   HSPI_HOST, &buscfg, 0); \\ 0 -no dma
   |                                  ^~~~~~~~~
esp32_spi.cpp: In member function void EspSpi::end()
 
esp32_spi.cpp: 37:25: error: 'SPI3_HOST' was not declared in this scope; did you mean 'SPI2_HOST'?
   37 |     #define VSPI_HOST   SPI3_HOST
   |                         ^~~~~~~~~
esp32_spi.cpp:120: note  in expansion of macro VSPI_HOST
   HSPI_HOST)
   |                            ^~~~~~~~~
esp32_spi.cpp: In member function void EspSpi::start()
 
esp32_spi.cpp: 37:25: error: 'SPI3_HOST' was not declared in this scope; did you mean 'SPI2_HOST'?
   37 |     #define VSPI_HOST   SPI3_HOST
   |                         ^~~~~~~~~
esp32_spi.cpp:146: note  in expansion of macro VSPI_HOST
   HSPI_HOST, &devcfg, &m_spi)
   |                                      ^~~~~~~~~
Error compiling libraries
```

**Expected behavior**
Compilation without errors

**Please complete the following information:**
 - library version: 1.1.5
 - LCD display type: SSD1306_128x32_I2C 
 - OS [e.g. linux, windows]: windows
 - Platform [e.g. Atmega328p, esp32, etc.]: esp32-c3
 - IDE if using some: Visual Studio + Visual Micro / Arduino 2 IDE

matt22free, 2024-10-04:
I have the same issue with ESP32C3 super mini

pdaripper, 2025-03-09:
строка esp32_spi.cpp: 37     #define VSPI_HOST   SPI3_HOST
нужно исправить на     #define VSPI_HOST   SPI2_HOST

lexus2k, 2025-03-28:
Please, find the change attached to this ticket