lexus2k/lcdgfx #125: DisplayST7789 with negative colors

lukkass222, 2024-08-18:
Hello, I started testing the colors on the display7789, and I noticed that the colors were inverted.
To confirm if the problem was with the display, I tested other libraries.

with [tft_espi](https://github.com/Bodmer/TFT_eSPI), everything worked fine.

with the Adafruit-ST7735, the colors were inverted, but in Adafruit there is the function `.invertDisplay(bool)`, putting `tft.invertDisplay(false); ` fixes the problem.

from that I did:

```
 #define ST77XX_INVOFF 0x20
 #define ST77XX_INVON 0x21

 display.getInterface().start();
 display.getInterface().send(ST77XX_INVOFF);
 display.getInterface().stop();
```
It's not ideal but it works

lukkass222, 2024-08-18:
ok, the display starts in inverted mode.

`0x21, CMD_DELAY,  10,    // INVON (21h): Display Inversion On"`

lexus2k, 2024-08-18:
That's interesting.  All displays I had in my hands required inversion mode.
The possible solution is to add invertmode API for st7789 displays.