lexus2k/lcdgfx #133: CS/DC/reset pins on expander

rawrbw, 2025-01-22:
so this library worked no problem on my pi pico with Arduino ide and directly connected. But for a project I'm doing i need to use a mcp23s17 expander on spi to control the cs/DC/reset lines. With this setup and Library modified (exp2.digitalWrite())so that expander is controlling the lines mentioned, the display garbles but i can see one line drawn in middle of screen and things are happening/animating within that small line.  Guessing it's some timing issue ? Any clue how to fix?

lexus2k, 2025-01-22:
Hello.
Do you have any hardware logic analyzer to capture digital signals for both setups and to compare them.
For me it looks that this is display initialization issue.
Did you check pull up / pull down configuration for cs/dc pins?

rawrbw, 2025-01-23:
I have a logic analyzer, see screenshots. It seems that Expander version is running slower(Im not sure how your code decides SPI speed?)... and also on the expander version of the code I have setup code for more spi devices that seem like are interfering, which I will try and remove tonight to see if I can get a clean version of the expander version working before trying to introduce rest of circuit setup. 

Regarding cs/dc I think you mean if I have put my "EXP2" in front of pinMode, to set OUTPUT? or maybe not... please explain.

[image]

[image]

[image]

[image]

rawrbw, 2025-02-02:
I came to the conclusion that this would not work as there would always be som point where both cs pins

closed 2025-02-02 by rawrbw